SkyEmie/huawei-honor-unlock-bootloader #1: General questions

thebadseeds, 2019-06-01:
Hey!

First of all: thanks for your great work. This seems like the easiest way to bruteforce the bootloader code. 

Yet, it's not really working for me. I entered my IMEI and let it bruteforce my Honor 5X over night. Unfortunately it didn't find the right code. 

So just to understand your approach better: why do you think the the unlock code consists of multiples of the Luhn checksum added to a base number? I couldn't find any information about this online. Did it work for you or is it just an assumption?

Also, before May 2018 Huawei asked for the IMEI and the serial number. Do you have any idea how to use both of this informations to improve your algorithm?

Thanks a lot!

advaithm, 2020-01-08:
I believe that huawei wanted imei number and serial to ensure that they were legitimate requests and/or to check the region and then a region based imei number.

SkyEmie, 2020-01-09:
This is the method used to generate the bootloader codes at huawei/honor.
I know you have to use the serial number, too, but I don't know how. I couldn't find anything about it, the only way is brute force. I've seen phones reboot every 5 failed attempts, whether they are from europe, asia or global, is that your case?

CsharpKoder, 2020-01-10:
@SkyEmie : Hi and thanks for your project
I have been running this for almost 24 hours and still can't find any match, was your solution tested with a specific huawei mobile ? which model ? how long did it take to find a match ?
What is your source of information for the way you calculate the possible unlock code ?

3c71, 2020-01-12:
Anyone successfully unlocked their device with this method?

CsharpKoder, 2020-01-12:
I can confirm it doesn't work in my case, I tested on 2 huawei phones.

3c71, 2020-01-12:
Well, it sucks as this might be the only hope to get root. 

I'm currently running this on a M3 Lite WiFi tab, so using the serial as IMEI. Very unlikely to work.

3c71, 2020-01-12:
It's been 5 hours, and it did only 1%... total process would be 20 days. Not sure it's worth it now.

CsharpKoder, 2020-01-13:
I ran it on 2 devices, starting with the minimum value until the generated code passed to 17 digits, at that moment I realized it's not working, took about 2 days for each device.

Baha2Odeh, 2020-01-30:
I tried it from 1000000000000000 to 10014530201474424 and it did not work :( it seems like luhn_checksum not used with Huawei phones

Nimbusman, 2020-01-30:
when the program asks "type imei digit" what imei should i input if i have 2? i have a huawei p20 lite, to most people it seems it doesnt work, when the program reboots the phone it boots in fastboot&rescuemode with frp unlock

dragonpro723, 2020-02-01:
any news how to avoid reboot after 5th attempt?

SkyEmie, 2020-02-02:
@CsharpKoder I've try with an Honor 5x, 8x, P10 lite and Huawei Y6, but with another Honor 8x it doesn't work, idk why.
@3c71 Hi, I don't know if i'ts method works with M3 Lite WiFi tab :/
Some devices reboot after 5 fails, I don't have solution except trying to set manufactury mode :/

dragonpro723, 2020-02-02:
Thanks tho for your work Skyemie, hope you find a new method one day. awesome work though!

droidacts, 2020-03-08:
Sadly, I've to report that this method didn't work on my Honor5x. I have adopted the codes for linux and have tried both imei (dual SIM). If I'm not mistaken, the trial unlock codes are simply multiples of sqrt(imei)*1024 + 10^15 since the Luhn checksum for a valid imei is zero. It took about 10 hours to go through one round (pass 10^16).

CsharpKoder, 2020-03-08:
It doesn't work, don't waste your time please.

robson-sz, 2020-03-24:
@SkyEmie Unfortunately `luhn_checksum` in your script gives incorrect output - it's easy to check when you have 15-digit IMEI number (15th digit is checksum calculated with Luhn algorithm). 
I've corrected it in my fork and will make a pull request as soon as my friends' honor 8 will be unlocked.
But first I must ask for confirmation: do I understand correctly that formula for unlock code is identical to this one?:
`1000000000000000+int(checksum+math.sqrt(imei)*1024) *x`
, where:
`checksum` is single digit
`imei` is 14-digit code
`x`-unknown number, script iterate starting from 0 till code is found

paulplayer, 2020-03-31:
> @SkyEmie Unfortunately `luhn_checksum` in your script gives incorrect output - it's easy to check when you have 15-digit IMEI number (15th digit is checksum calculated with Luhn algorithm).
> I've corrected it in my fork and will make a pull request as soon as my friends' honor 8 will be unlocked.
> But first I must ask for confirmation: do I understand correctly that formula for unlock code is identical to this one?:
> `1000000000000000+int(checksum+math.sqrt(imei)*1024)*x`
> , where:
> `checksum` is single digit
> `imei` is 14-digit code
> `x`-unknown number, script iterate starting from 0 till code is found
> 
> Edit: unnecessary space in formula
So have you succeed on any huawei phone?

robson-sz, 2020-04-01:
@paulplayer No, sadly it didn't work.
I've tried both IMEI numbers, each one for about 10h - script generated numbers bigger than `10000000000000000` (starting number *10) and i've stopped it.
My fork is public but still not working for me, you can try it.
Still hope for @SkyEmie response -  maybe we can figure it out together.

GuillaumeBroggi, 2020-04-02:
Thanks for the work!

@SkyEmie @robson-sz I have tested both your approaches with a Mate 8 ((NXT-L29), it did o't manage to find the unlock code.

paulplayer, 2020-04-02:
@SkyEmie  @robson-sz It seems like the brute force part not right,  I've cracked my honor by third party service, they offers me my BL code, it doesn't fit in 
> 1000000000000000+int(checksum+math.sqrt(imei)*1024)*x
But my BL code is a 16 long number. May be just brute force by a dictionary

robson-sz, 2020-04-02:
@paulplayer:Congratulations!
How long did it take?
Would you mind sharing your IMEI and BL code so we can try to look for pattern?
Btw 1000000000000000 is also 16-digit long.

paulplayer, 2020-04-02:
less than 1 min, it's an online service by mapping my usb port to a remote pc, so I just don't know how he done this. Here is my info, wish can help u.
An old model phone, huawei honor 7i ATH-AL00, IMEI: 863546031150590, BL: 3155366070293801.

robson-sz, 2020-04-02:
Thank you, will look into.

bbb0, 2020-04-13:
For me it could not find the correct code. Honor 8 (FRD-L09). Tried with the first IMEI and with the second IMEI too, it took about 9 hours to complete.
Maybe the fastboot throws another message when it works? When I reboot the phone with the fastboot reboot command, it says OKAY, not SUCCESS. Maybe when it get the correct code, it says the same? Or the correct code is not within the searched region (we try only about 10^16/30000 numbers).
I wonder where this increment equation come from? int(luhn_checksum(imei)+math.sqrt(imei)*1024) as increment in each step, but how did you guess this? At the square root we will get a float number and it is not equal how we convert it to integer (floor / ceil). Also why we start at 1000000000000000 and not at 0000000000000000?

From the numbers we see in paulplayer's comment it is clear that this bruteforce method could not work on that phone.
(3155366070293801 - 1000000000000000) / 30091421421 = 71627.2601 steps, which means we could never get this number. TBH it does not matter if you try this with ceil or floor, won't get integer number after division.

dezechristophe, 2020-04-29:
@SkyEmie do you know someone who successed to unlock his phone with your script ?

dezechristophe, 2020-04-29:
so it didn't work for me !
I went until 10002558807587722
honor 5x

35609902357, 2020-04-30:
I ran the script for a whole month, I got a message like `FAILED: wrong password` and an indication of 0.009s for each attempt, or something like that. After one month it was still running. I see some users here indicated a range of codes they tried, no codes were shown to me, just the wrong password thing. I enabled OEM unlock in settings and ran the script, which rebooted into fastboot mode and then I gave confirmation to start the process. Did I skip something?

bbb0, 2020-04-30:
>  Did I skip something?

You did it right. But as I stated before, the script does not check all passwords, if it would check all the script should run for 30000x more time (in my case it was running about 4hours to check all 16digits long numbers with the given increment). If I let it run until it checks all possible numbers, that would take about 12000 hours, which is 500 days :)
So as long the correct equitation is not known, the whole process is pure luck based.

35609902357, 2020-04-30:
@bbb0  So if I leave it running for 500 days (worst case scenario) I'm guaranteed ~a dead battery~ an unlocked bootloader? Or can it still fail after 500 days?

bbb0, 2020-04-30:
Only if you replace the increment (which is counted from the IMEI you enter) by one. So if only increment the numbers by one (try out all cases from 1000000000000000-9999999999999999) you should get an unlocked bootloader.

closed 2020-05-15 by SkyEmie