SkyEmie/huawei-honor-unlock-bootloader #2: you just used luhn_checksum once!

apotox, 2019-12-08:
luhn_checksum of imei is always a 0 its like the isbn for books

SkyEmie, 2020-01-09:
Yeah but, no, it's start with 11 base length character

tburschka, 2020-04-21:
@SkyEmie I'm not familiar with python, but are you sure `int` is in python always a 32 bit value? (this would explain the (11 base characters you've mentioned), but i don't think so. If it is, how is the 16 character long value converted? On my machines (windows/linux) it seems that an int a 64 bit variant:

    Python 3.8.2 (tags/v3.8.2:7b3ab59, Feb 25 2020, 22:45:29) [MSC v.1916 32 bit (Intel)] on win32
    Type "help", "copyright", "credits" or "license" for more information.
    >>> print(int(861794042848958))
    861794042848958
    >>>

closed 2020-05-15 by SkyEmie