SkyEmie/huawei-honor-unlock-bootloader #5: Any ideas for wifi tablets?

nayavu, 2020-01-12:
I have MediaPad T5 AGS2-W09 (wifi) which obviously does not have IMEI, but has MAC instead.
Any ideas how to adopt yours brute force method for wifi devices? The only thing I can think of is to convert MAC into a long int and to try the same algorithm but not sure if it makes any sense.

3c71, 2020-01-12:
Same here, got a M3 Lite BAH-W09 (wifi) with no IMEI.

3c71, 2020-01-12:
Here it said that Antutu was providing the IMEI on wifi tablets. So I tried and got a PFSNU number, the same one I gave Huawei a couple of years ago asking for the unlock code (which they provided but didn't work at all).

This is the same ID I get when connected to ADB, and the same ID that can be found in about tablet, state, serial number.

However it's only 11 digits, after the 'PFSNU'. So I'll be trying to use those adding 3 leading 0s to make it 15 digits. 

It's running now, no idea how long it'll take or if it will succeed. Worth a shot though.

nayavu, 2020-01-16:
Any luck with it?

3c71, 2020-01-17:
It did 1% in 5 hours, so I gave up as no-one appears to confirm it works.

dcosta72, 2020-01-20:
any progress on this? I'm on the same boat on a T3 AGS-W09...

SkyEmie, 2020-01-22:
For tablets, I don't know if the bootloader is the same like phone I can't test this :/

closed 2020-05-15 by SkyEmie