SkyEmie/huawei-honor-unlock-bootloader #6: Avoiding 5 attempts 30s delay

eddleloup, 2020-01-16:
Hi, in my case (with a Huawei Mate 10 pro) the phone would reboot into system after 5 attempts requiring me to reboot with adb manually, so I made a minor change, rebooting into fastboot every 4 unsuccessful attempt.  Bellow the code 

```python
n=0
    while(unlock == False):
        n+=1
        sdrout = str(os.system('fastboot oem unlock '+str(algoOEMcode)))
        sdrout = sdrout.split(' ')
        for i in sdrout:
            if i == 'success':
                return(algoOEMcode)
        if n%4==0:
            n=0
            os.system('fastboot reboot bootloader')

        algoOEMcode = algoIncrementChecksum(algoOEMcode, checksum)
```

SkyEmie, 2020-01-22:
Hi,
To avoid reboot after 5 failures, try to set your phone in manufactory mod, type in dial *#*#2846579#*#* and in background settings set manufactory mod.
 If it doesn't work, I don't have solution to fix this problem :/

dragonpro723, 2020-02-01:
got my nexus a few days ago and hate I cannot unlock. I have sky's version. How can I make it so it can bruteforce without rebooting every 5th time

dragonpro723, 2020-02-01:
very nice work tho skyemie, hope you might find a solution for this.. really hate I cannot unlock the bootloader

SkyEmie, 2020-02-02:
I've made this script in theory it works, but if your bootloader reboot every 5 fails I unfortunately have.. no solution, (Btw I don't have this problem with my honor 8x, but others 8x have this "problem" too so I don't understand what's going on :/ )

kur0tenh, 2020-02-12:
Trying this out on a regular Mate 20 and used @eddleloup fix to circunvent the 30S reboot issue, I tried getting into manufacture mode with *#*#2846579#*#* but it defaults back to HiSuite mode all the time. My guess is that it could be a behavior coded at the ROM variant specific to my service provider which is kind of a bummer since force rebooting every 4 tries kinda puts a lot strain to the device as I'm detecting it to get toasty after a couple of reboots, not to mention that it takes a little longer.

I made some changes to the code in order to retrieve every generated key in order to be able to stop and continue later since I have to use the device on a regular basis, so I introduced an input:

last_code = int(input('Type last code used, if first time then use 1000000000000000 \n'))

and made that variable:

def tryUnlockBootloader(checksum, last_code):
    unlock      = False
    # algoOEMcode = 1000000000000000 #base
    algoOEMcode = last_code
.
.
.

my other take to avoid rebooting every 5th failed try would be to try to sideload other official ROMs looking for one that doesn't default back to HiSuite every single time (but that's just a theory).

WolfganP, 2020-02-13:
> Trying this out on a regular Mate 20 and used @eddleloup fix to circunvent the 30S reboot issue, I tried getting into manufacture mode with `*#*#2846579#*#*` but it defaults back to HiSuite mode all the time. 

Same here. I tried the same with a P20 lite, and it seems the Manufacturer mode doesn't stick, as it doesn't survive the reboot initiated by the program and falls back to the 5 tries only.

joerg65, 2020-02-16:
I also tried @eddleloup fix since about 10 days on a P30 lite. Now I modified the code to count the maximum attempts until overflow. I get 3284424 attempts that multiplied by 12s each attemp results in 456 days ...
I think I will stop it now, as it makes no sense to wait and at the end to realize that it is not working...

afsal98, 2020-02-18:
how much times it will try to unlock. i waited for around 30 mins.but it doesn't completed. can anyone say the average time to taken.

afsal98, 2020-02-18:
which imei code is suitable for unlock.dual sim phone have 2 imei codes.pls reply.

joerg65, 2020-02-19:
@afsal98: I don't know which IMEI. As I wrote I gave up on this. My phone (P30 Lite) delayed each fastbooot password probe by 11.5sec and with @eddleloup fix it rebooted after 4 attempts again into fastboot mode. After aprox 3 days the battery was drained and the phone booted into Android and charged. Then I had to give the command adb reboot bootloader by myself. The script waited at this stated and went on when the phone again was in fastboot mode. And as I wrote above, the attemps until the password overflows are 3284424. You can calculate how much time this need.

kur0tenh, 2020-02-20:
> @afsal98: I don't know which IMEI. As I wrote I gave up on this. My phone (P30 Lite) delayed each fastbooot password probe by 11.5sec and with @eddleloup fix it rebooted after 4 attempts again into fastboot mode. After aprox 3 days the battery was drained and the phone booted into Android and charged. Then I had to give the command adb reboot bootloader by myself. The script waited at this state and went on when the phone again was in fastboot mode. And as I wrote above, the attemps until the password overflows are 3284424. You can calculate how much time this need.
> 
> I also tried to downgrade firmware without success.
> And I tried the Qu1ckr00t method CVE-2019-2215 without success.

I think that you may sideload another firmware from the same version rather than downgrading it.

MKCOOL142, 2020-05-09:
> which imei code is suitable for unlock.dual sim phone have 2 imei codes.pls reply.

Definitely use the imei1 code if you only have 1 card inserted and only if it's in sim1 slot, if you have only one sim but in sim2 slot then use IMEI 2, I can't confirm the sim 2 thing but I did some testing and my emui2 is reported as "invalid" when looked up at IMEI.info

Doping11, 2020-05-11:
> 
> 
> > which imei code is suitable for unlock.dual sim phone have 2 imei codes.pls reply.
> 
> Definitely use the imei1 code if you only have 1 card inserted and only if it's in sim1 slot, if you have only one sim but in sim2 slot then use IMEI 2, I can't confirm the sim 2 thing but I did some testing and my imei2 is reported as "invalid" when looked up at IMEI.info

Must a SIM card be inserted to unlock the bootloader?

AkgunFatih, 2020-05-12:
fucking Huawei
we pay for it but don't own it

Doping11, 2020-05-13:
Multiply your IMEI by this factor 3,633366873662295. The Result might be your unlock code.

I have divided my unlock code by my IMEI. Lets try it. You can´t lose anything....

AkgunFatih, 2020-05-14:
keep in mind that if you are using the latest android security patch
there is no way you can unlock your phone
says the guy from dc-unlocker

closed 2020-05-15 by SkyEmie

DangerousPony204, 2020-06-07:
If your phone doesnt reboot into bootloader after the first try, but just reboots back in android.
Make a 'example.bat' file with:
```
adb reboot-bootloader
example.bat
```
and run it along skyemies code
This way it will reboot in bootloader as soon as it is possible.
I dont know or this is going to work because i just figured this out myself and its running on my pc right now.