SkyEmie/huawei-honor-unlock-bootloader #8: Fastboot Mode

Jacksoft, 2020-02-05:
Hi, would be good to have an option to start the bruteforce directly if the device is in fastboot mode already.
I'll explain. I have an ALE-L21 stuck on bootloader logo. The only available thing I can do is to enter in fastboot, so the script "would not work" because no ADB is detected. For exable you could put an option like "Press 'F' if you're already in Fastboot Mode" after the IMEI input.
Now I've jumped all the checks and I'm trying to get the bootloader code.

PS: There's a typo in deBBug :P

bbb0, 2020-04-12:
If you are already in fastboot, the first section would not work (which reboots your phone to fastboot mode), but after that the bruteforcing will work as it intended.

SkyEmie, 2020-05-15:
> EDIT: Is possible to add a line that describe the attempt? For example: "Attemp number #1: Key tested 12345678"
Done! :)

closed 2020-05-15 by SkyEmie