SkyEmie/huawei-honor-unlock-bootloader #9: It doesn't work, don't waste your time please

CsharpKoder, 2020-03-08:
For new comers, don't waste your time or raise your electricity bills, this technique is not working.

The author doesn't provide any information about the methodology, and when tested by several users

here, it turned out to be "Not Working"

WaseemAlkurdi, 2020-03-12:
He doesn't need to, go read the code.

CsharpKoder, 2020-03-22:
Why do you assume that I didn't read the code, I think you should go and read
what other people also got from this.

I read the code and I tested it on 2 devices, it doesn't work.

closed 2020-03-22 by CsharpKoder

WaseemAlkurdi, 2020-03-25:
> The author doesn't provide any information about the methodology

That shows that you obviously hadn't read the code. The code is literally self-documenting.
Tip: this is the part you would want to read:
```
def algoIncrementChecksum(genOEMcode, checksum):
    genOEMcode+=int(checksum+math.sqrt(imei)*1024)
    return(genOEMcode)

def luhn_checksum(imei):
    def digits_of(n):
        return [int(d) for d in str(n)]
    digits = digits_of(imei)
    oddDigits = digits[-1::-2]
    evenDigits = digits[-2::-2]
    checksum = 0
    checksum += sum(oddDigits)
    for i in evenDigits:
        checksum += sum(digits_of(i*2))
    return checksum % 10
```
> I think you should go and read what other people also got from this.

Even if it worked for nobody, that doesn't give you the right to whine about the code and especially about the _author_ of the code:
> For new comers, don't **waste your time** or **raise your electricity bills**, this technique is not working.

Makes you look stupid.
If it doesn't work for you, either do more research and look for an explanation as to why this particular method didn't work for you (Huawei didn't manufacture your phones as limited editions just for you ... others must have had the same problem), or otherwise **respectfully** report that.

> I tried your code and it didn't work for me. Others have also reported this problem, which suggests that this isn't an issue on my end. Looking into the issue, it turns out that this method is unsupported on my devices. Perhaps you might want to update `README.md` with a list of tested/compatible devices or at least a note that it might not work for everybody?

> I tried your code and it didn't work for me. Others have also reported this problem, which suggests that this isn't an issue on my end. Looking into the issue, it turns out that there is an error in line NN in your implementation of the algorithm, which I have corrected using this patch:

See how much more respectful and less cringeworthy that one was?

CsharpKoder, 2020-03-25:
سد حلقك و كل خرا أحسنلك و لا تتفلسف زيادة
فعلاً إنك حمار وواطي

WaseemAlkurdi, 2020-03-25:
كل بئرٍ بما فيه ينصح, ولن أنزل لمستواك.
(من كان يؤمن بالله فليقل خيرًا أو ليصمت)

CsharpKoder, 2020-03-25:
يعني نازل تتفلسف و تبهدل فيي و تقلي غبي و بعدين عامل حالك بريء !

نازل تقلي إنتا ما قرأت الكود و تتخوث و أنا قرأت الكود و جربته و حاولت أعدل عليه كمان

و بعدين أنا سؤالي لكاتبة الكود كان عن مصدر الطريقة، مش الكود نفسه، يعني على إيش اعتمدت

لما جابت هاي الطريقة، أنا جربتها و غيري جربها و ما حدا اشتغل معاه اشي

فهمت يا حبيبي و لا لسا ؟

WaseemAlkurdi, 2020-03-25:
> يعني نازل تتفلسف و تبهدل فيي و تقلي غبي و بعدين عامل حالك بريء !

ما حكيتلك غبي. حكيتلك بتبيّن أهبل لما تروح تطخ على ناس ما إلها دخل بسبب عدم بحثك عن سبب المشكلة.
انت فعليا عملت التالي: البرنامج ما اشتغل, إذن العتب على المؤلف وعلى البرنامج. طيب المؤلف أخطأ بعدم وضع أسماء الأجهزة المتوافقة. هذا لا يبرر الإهانة (مقصودةً كانت أم غير مقصودة) كما في أول تعليق, بل الصواب كان (كما أوضحت لك) البحث عن أصل المشكلة لا انتظار الحل جاهزًا وإلقاء اللوم على الغير لعدم توفيره العلاج السحري لكل مشكلة.
> نازل تقلي إنتا ما قرأت الكود و تتخوث و أنا قرأت الكود و جربته و حاولت أعدل عليه كمان

لا أعلم الغيب ولا أتوقع أن غيري من البشر يعلمه. لو قلت هذا ووضّحت محاولاتك (ليستفيد منها المؤلف كي يعدل كوده) لكفيت نفسك شر التهمة! 😅️

> و بعدين أنا سؤالي لكاتبة الكود كان عن مصدر الطريقة، مش الكود نفسه، يعني على إيش اعتمدت

مما يثبت حدسي (لا أقول علمي, بل حدسي) بأنك لم تبحث عن أصل الطريقة, إذ أنك لو فعلت لرأيت أن هذا السكريبت ليس فريدًا من نوعه, بل يوجد الكثير من أشباهه:
https://github.com/WilliamWangyueming/bruteforce-bootloader-unlock
https://github.com/Martazza/Huawei-Bootloader-Unlocker
https://forum.xda-developers.com/huawei-p20-pro/help/brute-force-unlock-unlock-codes-t3826968
https://android.stackexchange.com/questions/220461/unlock-huawei-bootloader-is-brute-force-the-only-way-now

CsharpKoder, 2020-03-26:
قسماً بالله إنك إنسان وقح، و أنا صادفت في حياتي كثير من أمثالك خصوصاً في مجال الآي تي
النوع اللي حتى لما يعرف إنو غلط، ما بيتراجع و لا بيعتذر

> انت فعليا عملت التالي

إذا كنت ما بتعلم الغيب، فليش حاطط حالك محامي دفاع و نازل تتفلسف و ترمي تهم على الناس ؟ أترك هاي العقلية المريضة و خاطب الناس بتواضع و لا تتصنع العلم و المعرفة.

لعنة الله عليك فعلاً