SkyEmie/huawei-honor-unlock-bootloader #10: How many passes does the program run and.... please read. AI implement???

ImKhizy, 2020-03-11:
i was wondering how many combinations is the program gonna try and did you write some code to exclude/delay i.e put forward in the list the more stupid combos like 121212, ABABABAB,AD9AD9AD9, 11112222, AAAADDDD OR AADDAADD PR AAADDDAAADDD so on and so forth. if I understand properly not digits are being replaced instead only some are being replaced, doesnt that allow implementation of linear regression if people could contribute their unlocked phones imei and unlock code and then we train a neural network to find a universal algorithm , even if the algorithm can confirm 75% of the digits we will only need to try 25% of the remaining combinations.

Liupaperbox, 2020-03-18:
it's a very good suggestion, but it's very tough to find people who can contribute their unlocked code.

ImKhizy, 2020-03-18:
Cant you set some like prize or like 20$ for one of 100 or 500 people.
Kinda sounds reasonable. We can always just not give the cash. It wouldn't
be stealing since we didn't take anything material from them in the first
place

b0ltun, 2020-03-30:
@ImKhizy @Liupaperbox @SkyEmie 
 i got some working bootloader codes (it`s 100% it`s working well)

one code i am providing for you with an IMEI

4RJW7PI5N19KTBN6
JK6H20LUH4GCSRFG - 864623031584190
41GJBPCK3Y5YJYYN

ImKhizy, 2020-03-31:
This is great but I need atleast 50 to filter coincidental sequences,
linear regression works best with 1000s of training data but I suppose we
could try it with 50 though.

tburschka, 2020-04-21:
@b0ltun so you got an unlock code based on alphanumeric values. That's interesting, since the code seems to be only testing numerice codes...

SkyEmie, 2020-04-27:
Be careful, because on most European honor and huawei models, the bootloader code consists of numbers only. Exception for some tablets too

tburschka, 2020-04-29:
> Be careful, because on most European honor and huawei models, the bootloader code consists of numbers only. Exception for some tablets too

that's the crux, i'm on a chinese variant. @SkyEmie where did you get your information about the bootloader codes?

Doping11, 2020-05-12:
> 
> This is great but I need atleast 50 to filter coincidental sequences, linear regression works best with 1000s of training data but I suppose we could try it with 50 though.
> […](#)

I can provide my data (imei, serial nr, unlock code) of my Honor X6.
Can I send it to you via E-Mail or something like that?

SkyEmie, 2020-05-15:
@tburschka From EMUI website when it was still online
And no we can't do IA on this data, it's absolutely random.

The only 2 ways today that exist are: 
- Downgrade your version (android 10 to older builds as possible) and pay a tool like dcunlocker which will use a system vulnerability to get the hash of the bootloader code, and via a rainbow table, give the code 
- or bruteforce the code via my method, which may not work if your bootloader has been patched if it's not european

OnceDot, 2020-05-23:
Hi, fastboot mode is showing a prompt "unlock bootloader ?"

And also I suggest to change `input` to `raw_input` because the script output.

> Traceback (most recent call last):
  File "huawei-honor-unlock-bootloader-master/unlock.py", line 75, in <module>
    input('Press any key to reboot your device..\n')
  File "<string>", line 0
    ^
SyntaxError: unexpected EOF while parsing

closed 2020-08-16 by SkyEmie