SkyEmie/huawei-honor-unlock-bootloader #13: Is it going right for me?

j-brake, 2020-03-31:
Thanks for the code!
I have entered the IMEI and the program seems to be running, the following messages keep appearing:
FAILED (remote: 'Invalid key, unlock fail')
fastboot: error: command failed
FAILED (remote: 'Invalid key, unlock fail')
fastboot: error: command failed
.
.
.
It's been going on for about two days now, should I keep waiting? Or is it useless?
By the way, I have a Honor 5x

TG310, 2020-04-13:
Does the messages appear slow and consistently or it is spamming rapidly?

j-brake, 2020-04-13:
I would say the messages spam fast and consistently ;-)

TG310, 2020-04-14:
I think you are on the right way.
but 2 days straight is enough time to run through all the possible numbers
(I had only ran it for 12 hours but its already half the way)
I think its useless for you now.

If you want to try your luck then why don't you try this:
https://github.com/Martazza/Huawei-Bootloader-Unlocker

closed 2020-05-15 by SkyEmie