SkyEmie/huawei-honor-unlock-bootloader #14: FAILED (remote: 'check password failed!') fastboot: error: Command failed

throwoutofcoffeeexception, 2020-04-01:
Hello i get the error: 
FAILED (remote: 'check password failed!')
fastboot: error: Command failed
What is the problem?

ArcherN9, 2020-05-08:
I incurred the same. Its weird because fastboot is on the path and I can run it from any folder. 
[image]

ArcherN9, 2020-05-08:
I just realized, this may just be for Windows. It doesn't work on a Mac, does it?

grossmaul, 2020-05-08:
Yes, its a script for Windows. Mac needs other system calls (e.g. you need a `./` infront of the program you want to call if its in the same directory) and you have to install adb and fastboot on another way. I recommend to google "adb fastboot install macos".

Here adb and fastboot are `.exe` files which only work on windows.

ArcherN9, 2020-05-08:
I understand that. I checked the `unlock.py`. It should work for Linux / Mac all the same provided `adb` and `fastboot` are added on the PATH which they are for me.

closed 2020-05-15 by SkyEmie

emha-koeln, 2020-07-26:
Hi,
read [README.md](https://github.com/SkyEmie/huawei-honor-unlock-bootloader/blob/master/README.md)
"- **My phone reboot every 5 failed attempts**
Sorry.. I've seen this in a few people and I have no idea. Its look like an additional protection. It happens on some models, having the same version, identical."

AFlyingBird90, 2020-07-27:
> Hello i get the error:
> FAILED (remote: 'check password failed!')
> fastboot: error: Command failed
> What is the problem?

I came across the same problem on a Huawei Nova youth (P10 lite). It shows in terminal (macOS):
"
Press any key when your device is ready.. (This may take time, depending on your cpu/serial port)

sh: title: command not found
                                                   FAILED (remote: 'check password failed!')
fastboot: error: Command failed
sh: title: command not found
"

It just stops there. What does "sh: title: command not found" mean?

It seems not to be the same problem in https://github.com/SkyEmie/huawei-honor-unlock-bootloader/issues/37, where it shows:

"
FAILED (remote: 'check password failed!')
fastboot: error: Command failed
FAILED (remote: 'check password failed!')
fastboot: error: Command failed
FAILED (remote: 'check password failed!')
fastboot: error: Command failed
FAILED (remote: 'check password failed!')
fastboot: error: Command failed
Rebooting into bootloader OKAY [ 0.021s]
Finished. Total time: 0.023s
"

emha-koeln, 2020-07-27:
Hi,
>sh: title: command not found

the shell says, there is no command 'title'
'title' is for [Windows](https://docs.microsoft.com/en-us/windows-server/administration/windows-commands/title_1)...
Are you on a none Windows-system?