SkyEmie/huawei-honor-unlock-bootloader #21: Lenght of IMEI

RainbowSimon, 2020-04-30:
Hey, I tried to run it with my Honor 6x and since you tested it with a Honor 5x and 7x 6x should work too.
However I started filling in my (first) IMEI and let it run from the start 1000000000000000. I assume it should be running til 9999999999999999 however it's at 10231571622834406 already. Did I do anything wrong or should it do much longer?

SkyEmie, 2020-05-15:
Your phone isn't connected to ADB

mmorais-idealsumario, 2020-06-01:
Not trying to hijack this thread as I think this is related, but what is the reasoning behind the code generation attempts: `genOEMcode+=int(checksum+math.sqrt(imei)*1024)` ?
I did a rudimentary web search and couldn't find anything. I'm curious as well as interested in knowing if this would apply in the phone model I have. Thanks for the good work :)

closed 2020-08-16 by SkyEmie