SkyEmie/huawei-honor-unlock-bootloader #22: Doesn't work for Honor 5X

grossmaul, 2020-04-30:
I saw that you already tested it for the Honor 5X, so I tried it with my 5X and let it run the full length of the 16 digit numbers of unlock codes (about 8-9 hours). But now the skript started with the 17 digit numbers and the phone is still not unlocked...

I googled around already and many people tell that the unlock is actually an alphanumeric code. May you have to add all letters or change to hex numbers?

SkyEmie, 2020-05-15:
Hi, I could only test on European models (as indicated in the readme) and yep, the bootloaders of others regions are different, and I have no idea how the ones in the other regions work

grossmaul, 2020-05-15:
Yeah, I have an European version too, so I think the bootloader could be the same.

The phone was bought in Germany.

iHeadRu, 2020-05-16:
Honor 5C (Russia). Also no luck in 16 digits.

Doping11, 2020-05-17:
> 
> 
> Yeah, I have an European version too, so I think the bootloader could be the same.
> 
> The phone was bought in Germany.

I have an Honor 6X (german/european Version/ model nr.: BLN-L21) and unfortunately the bruteforce-script didn´t find the correct unlock code. The phone was on Android 7 / EMUI 5.0.3

@grossmaul When I read your nickname, I knew you were from germany.

grossmaul, 2020-06-18:
Okay, now I decided to use the DC-Unlocker, which worked for me.

The Unlock code for the Honor 5X was 1450034138xxxxxx (all numeric).
Since I already ran your unlocker code I can say that your algorith didnt hit my unlock code. Can you tell why you have chosen the Luhn algorith here? At least for me it didnt work with that.

closed 2020-08-16 by SkyEmie