SkyEmie/huawei-honor-unlock-bootloader #26: P20 pro support

doronmak, 2020-05-22:
Is anyone tried this on the clt l09?

mschabhuettl, 2020-06-02:
I've tried it for about 10% of the possible combinations (~2 days of calculating bootloader codes). Meanwhile I researched into this topic stumbling on several Telegram groups. The admins told me the bootloader code is completely random (and could be alphanumeric). It also seems that the unlock code has nothing to do with your IMEI number. The only way to unlock your Huawei P20 Pro bootloader is by using the BLK-1 method: you need to apply a custom **B**oot**L**oader**K**ey to your device. BLK-1 costs ~40 euros. I am currently pending on their service, so I can't tell you (yet) if it works, but I'll keep you up to date (if i don't forget).

AAAFFF, 2020-06-06:
Hi @doronmak, I've tried it without succes: the phone reboote every 5 attempts! @mschabhuettl does your p20 pro did the same? And where did you find the BLK-1 method? I've never heard it

mschabhuettl, 2020-06-06:
> Hi @doronmak, I've tried it without succes: the phone reboote every 5 attempts! @mschabhuettl does your p20 pro did the same? And where did you find the BLK-1 method? I've never heard it

@AAAFFF Yes, my phone also reboots every 5 attemps. You have to apply the workarround described in [Issue #6](https://github.com/SkyEmie/huawei-honor-unlock-bootloader/issues/6#issue-550845717).  It basically increases a variable by 1 counting up. After it tried 4 attempts (failing), it forces fastboot to reboot the device and resets the counter to zero.
In case I try another attempt in brute-forcing the unlock code, I'll fix the code and submit a pull-request to @SkyEmie.

2nd: I don't know anymore how I discovered the Telegram-Groups, but if you're interested PM me and I'll provide you with a join-link.

AAAFFF, 2020-06-06:
Thank you so much @mschabhuettl ! I'll try the method with the workaround and if it won't works I'll text you privately.

mschabhuettl, 2020-06-06:
@AAAFFF You're welcome. You can clone my [forked version](https://github.com/mschabhuettl/huawei-honor-unlock-bootloader) - I implemented all the features you need.

cburlacu, 2020-06-24:
Did anyone succeeded to unlock the bootloader of P20 pro? (either using this script or that BLK method)

yerbestpal, 2020-06-26:
@cburlacu I am currently trying using the original script and the discussed forked script. Neither seem to be working: 'FAILED (remote: 'check password failed!')'.

yerbestpal, 2020-06-28:
@cburlacu Let it fun for a few hours with no success. I wasn't willing to continue because I was concerned about the temperature of my phone.

closed 2020-08-16 by SkyEmie