SkyEmie/huawei-honor-unlock-bootloader #27: Y6 2018

flydr2, 2020-05-24:
Just to let everyone know.

After about 20 hours the count finally arrived at 17 digits... With no luck.

The script works as it should..

My version is for North America so my guess it requires an alphanumeric code.

Thanks for the work on this..

lokkiii, 2020-06-24:
@flydr2   Kindly let us know in case you manage to find a way to try the alphanumeric combinations(of only 16digits) for the code.  It is very clear that many phones may have combinations alphabet and numbers, and many have put out their unlock code a bunch of times for the more recent models.   

I am a noob in python, and when I try to find a way to use combinations of 0-9 and a-z , I am unable to find any way to do it.  If you can pls direct us to a stackoverflow blog or a direct sample of your code, many would be grateful for that.    

SkyEmie  doesnt seem to respond since month of May I guess, and I am not sure all the people trying this code, are aware of the alphanumeric combinations required for recent phones.  So again, my request to anybody, is there a code that can make a systematic combination of length 16, with 0-9 with a-z....??

flydr2, 2020-06-24:
If your combination is letters and numbers then there is no way to find
it...

closed 2020-08-16 by SkyEmie