SkyEmie/huawei-honor-unlock-bootloader #28: Error

pApY-Dalix, 2020-05-25:
Hello, I'm trying the script on my Honor 8X, after following the instructions given, an error occurs after restarting the smartphone... Here's what the terminal looks like :
```
                                                   FAILED (remote: 'invalid command')
fastboot: error: Command failed
                                                   FAILED (remote: 'invalid command')
fastboot: error: Command failed
...
```
Can you tell me where the error came from, thank you very much.

pApY-Dalix, 2020-05-25:
After a new attempt, the terminal to indicate this in addition to the previous error :
```
Traceback (most recent call last):
  File "unlock.py", line 79, in <module>
    codeOEM = tryUnlockBootloader(checksum)
  File "unlock.py", line 24, in tryUnlockBootloader
    sdrout = str(os.system('fastboot oem unlock '+str(algoOEMcode)))
```

nazariiixa, 2020-05-27:
I have the same problem "fastboot: error: Command failed
                                                                                  FAILED (remote: 'invalid command')" for all atempt, with my Honor view 20

pApY-Dalix, 2020-05-28:
I think the error comes from the ADB and not from the python script, or a mix between the two :/

Jok00er, 2020-06-05:
I have the same problem with my Huawei y9 prime 2019. My computer is running on windows 10

24kcsplus, 2020-06-26:
I have the same problem with my honor 9x China version......

Remty5, 2020-07-04:
"fastboot: error: Command failed FAILED (remote: 'invalid command')"
Is absolutely normal, that's the output of fastboot when the unlock code is wrong, that's why this program searches for it, it should find it eventually.

closed 2020-08-16 by SkyEmie