SkyEmie/huawei-honor-unlock-bootloader #32: Requesting Features based on Error

lokkiii, 2020-06-05:
Dear SkyEmie,

###  You are brilliant to create this code here, as it will help so many. And today when I came across, and started testing, I got the following error after the code  tried about 2billion combinations: 

(bootloader)  Navigate to Developer options, and enable "OEM unlock"!
FAILED (remote: 'Command not allowed')
fastboot: error: Command failed
                                                   (bootloader)  Necessary to unlock FRP!

###  Before that  it would only mention "fastboot: error: Command failed      FAILED (remote: 'invalid command')"

So that means the code is working, but then it stops when the phone gets overheated or it stops communicating, I guess.   

So can you please add a code in the script where it can make a note of the last combination tried, 
    and then wen I reattach the phone after a day or few (and type in the same IMEI number as previously entered), it will recognise the IMEI from the log, and continue from where it left off on combination previously tried, rather than again restarting from the beginning of 10000000000000....

That would also be mighty helpful for people who may have to split time for the unlocking.  In my case, when I again restarted my phone, I found that the OEM unlocking and the USB debugging is all enabled. But the phone was burning hot. And probably some USB wire issue.

lokkiii, 2020-06-06:
Dear SkyEmie.. Also can you please have a look at the article in this:
https://android.stackexchange.com/questions/225683/bootloader-code-for-android-devices

In the comments section, you will find Andy, who states there is no way of unlocking EMUI10 at all.  Can you plss plssss take a few mins to check if that is authentic, or if he is just discouraging me from going further, because he is from that country?

mschabhuettl, 2020-06-08:
> Dear SkyEmie.. Also can you please have a look at the article in this:
> https://android.stackexchange.com/questions/225683/bootloader-code-for-android-devices
> 
> In the comments section, you will find Andy, who states there is no way of unlocking EMUI10 at all. Can you plss plssss take a few mins to check if that is authentic, or if he is just discouraging me from going further, because he is from that country?
> 
> (btw I have a Huawei p20 lite, on which you have already tested :) )

@lokkiii Can confirm this. EMUI10 will completly lock down the ability to unlock your bootloader.
Did you manage to unlock the bootloader on your Huawei P20 Lite?

lokkiii, 2020-06-09:
> Did you manage to unlock the bootloader on your Huawei P20 Lite?

No, it keeps going on forever.. and it heats up my phone every 1hr or so, and then stops. Hence I was wondering if there is a way to know, what Andy says in that blog, is true.

mschabhuettl, 2020-06-09:
> > Did you manage to unlock the bootloader on your Huawei P20 Lite?
> 
> @mschabhuettl Thanks for getting on to this. And no, it keeps going on forever.. and it heats up my phone every 1hr or so, and then stops. Hence I was wondering if there is a way to know, what Andy says in that blog, is true.
> 
> And I stand corrected on the model.. In settings I found this...
> Model: HRY-AL00Ta (But its the same afaik, as Huawei P20 Lite for other countries)
> EMUI: 10.0.0
> 
> Hoping it will still work

In terms of temperature i used an old PC-Fan to cool my phone.
Until now i got up to `last_attempt = 1626788393545591` - with my bootloader still locked on my Huawei P20 Pro (European).
@SkyEmie No offence, but did you successfully unlock any bootloader using this method, on your devices?

lokkiii, 2020-06-11:
@SkyEmie Pls can you modify the program to be alpha-numeric, because in many places there are hints that for a few phones, some alphabets are to be seen in the unlock code, and I have a feeling I may be among them lucky lot. Currently, only numbers are being tried and I have moved to almost an 18digit code, starting from the 16digit that it starts by default... Hopes are dim, that it will only be a number with no characters involved...

fwSmit, 2020-08-16:
Hi all,
My honor 10 was also kindly updated by Huawei to EMUI 10. This had as result that this script could really quickly test many unlock codes without much delay. This is probably because the unlock command was being rejected. 
It is however possible to downgrade the EMUI version to 9 by using their software HiSuite. THIS WILL REMOVE ALL DATA ON THE PHONE!
I hope this helps you in your journey of unlocking your phones. I still haven't succeeded either, so I wish you luck!

SkyEmie, 2021-01-12:
Hi,
Andy is right, if you have upgraded via to the latest version of android, this is no longer possible because the bootloader is patch. You will have to downgrade. (don't forget to make a backup before!)

@lokkiii I won't make this modification because if your bootloader is composed with non-numeric characters, you can leave the script running for years... Hoping that your bootloader according to your model and your region is composed with numeric number, otherwise it will be impossible to bruteforce :/

closed 2022-05-26 by SkyEmie