SkyEmie/huawei-honor-unlock-bootloader #34: How long will it take to get the right code?

nICEnnnnnnnLee, 2020-06-08:
Write the time you spent pls...
I'm considering whether to buy a code or not.

nICEnnnnnnnLee, 2020-06-11:
Well, I give it up...
I have tried algoOEMcode from 1000000000000000(16 digits) to 10078719127057140(17 digits), unfortunately I didn't hit the code.

nICEnnnnnnnLee, 2020-06-13:
I'm not sure whether the head and every step of brute force is appropriate,  
but the right algoOEMcode of mine just begin with 0 and less than 1000000000000000 since it was unlocked using command: 
```
fastboot oem unlock 060082931962**** (use * instead for privacy)
```
BTW, my device : Honor 7(PLK-TL01H)

lokkiii, 2020-06-13:
@nICEnnnnnnnLee that is awesome news. Congrats!  Btw, did you use the same code that SkyEmie had provided, or have you used another code of your own pls. Kindly let us know as many are struggling since 1 or 2yrs to find a break in this matter...

Thank you...

David-H-1, 2020-08-08:
Newer devices use 16 alphanumeric codes which is 36^16=7958661109946400884391936 possibilities. With some models rebooting every 5 failed attempts, it will take decades to unlock newer models.

closed 2020-08-16 by SkyEmie