SkyEmie/huawei-honor-unlock-bootloader #35: Error messages "Navigate to developer options, and enable OEM unlock"

bantour, 2020-06-10:
Is this the normal operation of the program? Is this how it should be?
[image]

Jppx, 2020-06-27:
This seems to happen after around an hour of search through the permutations. I have to reboot the phone to normal, then start the process again at the latest code. (If it happen right away, you maybe dont have the OEM Unlock option enabled on the handset. (System & Updates >> Developer Options >> Enable OEM Unlock)

bantour, 2020-06-27:
I get this message immediately when I run the script. Thе option "Enable OEM Unlock" is enabled.

Jppx, 2020-06-27:
I only get that message after around 1 hour in fastboot mode. During that hour I get the standard ' FAILED (remote: 'invalid command') repeated for every attept at the code.
On your phone screen in FastBoot Mode does the 'FRP Unlock' message in red appear?

bantour, 2020-06-27:
[image]

closed 2020-06-27 by bantour

Jppx, 2020-06-27:
You may have to take off the FRP lock then.  I think it involves removing the google account from the phone. Check online - I seen a procedure for this a while back.

Jppx, 2020-06-27:
I have PHONE Locked  (Green)
          FTP Unlock  (Red)