SkyEmie/huawei-honor-unlock-bootloader #37: Is the script working or not?

relex458, 2020-06-21:
**the following messages keep appearing rapidly, whilst the bootmenu is refreshing**:
                                                   FAILED (remote: 'check password failed!')
fastboot: error: Command failed
                                                   FAILED (remote: 'check password failed!')
fastboot: error: Command failed
                                                   FAILED (remote: 'check password failed!')
fastboot: error: Command failed
                                                   FAILED (remote: 'check password failed!')
fastboot: error: Command failed
Rebooting into bootloader                          OKAY [  0.021s]
Finished. Total time: 0.023s

I wanted to know if im doing anything wrong, since the guy who wrote #13 seems to have done it right, yet he gets another message then me.

relex458, 2020-06-21:
also: this was already asked (#14 ), but no one answered

yerbestpal, 2020-06-26:
Isn't this just the message displayed as it fails brute-force tries? I hope so, because I have had it running in this fashion for over an hour now.

Remty5, 2020-07-04:
Well yeah that's the output of the fastboot oem unlock command when you provide the wrong unlock code, which this program tries to find, so of course it's gonna try hundreds of them.
@relex458 please close thus issue.

emha-koeln, 2020-07-26:
Hi,
see [14](https://github.com/SkyEmie/huawei-honor-unlock-bootloader/issues/14)

closed 2020-08-16 by SkyEmie