SkyEmie/huawei-honor-unlock-bootloader #38: Mediapad T3 7 Didn't work

sasadara, 2020-06-24:
Anyone succeeded with the tool? I tried all 16 keys (17 length give length error). Including from 0 + str(100000000000000)

psolyca, 2020-10-12:
I am trying to unlock a Mediapad T3 7 wifi-only.
The problem is that this version of the tablet do not have any IMEI code only the serial number.
So I disable all the code to check and compute from IMEI and change the code to get `imei=1`

I do not know about others devices but what is the message everybody gets when cade is falling ?
I have
`FAILED (remote; 'Invalid key, unlock fail, error num: 8')`

I will also try all my Huawei later as I already have codes for 2 of them, G620S-L01 and EVA-L09 (P9)

SlowLogicBoy, 2021-01-01:
MediaTab T3 8 KOB-W09, same as @psolyca. Also same error when putting any random valid huawei imei. I changed `unknownfail = False`

closed 2022-05-26 by SkyEmie