SkyEmie/huawei-honor-unlock-bootloader #39: Doesnt Find Code for Honor View 10 (UK)

Jppx, 2020-06-27:
Model: Honor View 10 (BKL-L09)
EMUI: 10.0.0.177 (C432E4R1P4)  
Country: UK

Ran through all permutations based on the algorithm (including values below 15 digits). This took around 12 hours total.
**No unlock code found**. So it may be true that after EMUI 10, there is no way to unlock the bootloader on Hauwei phones.

TipzTeam, 2020-07-05:
I feel like downgrading to Oreo before the June patch should do the trick.

Jppx, 2020-07-05:
Can't downgrade from Android 10, because I cant unlock the bootloader.  - Catch 22.

TipzTeam, 2020-07-06:
No, you could use the SD downgrade method.

lokkiii, 2020-07-06:
Is there any tutorial you can point to? Probably on youtube, or can you please write down the steps and help, of how can we achieve it without the bootloader unlocked please?

TipzTeam, 2020-07-09:
> @TipzTeam Is there any tutorial you can point to? Probably on youtube, or can you please write down the steps and help, of how can we achieve it without the bootloader unlocked please?
> 

Searching online will be your answer.

llight3, 2021-03-09:
This is prabably because the EU region used alpha numeric, try the random approach

sohail4k, 2021-03-13:
Downgrading to EMUI 9 from higher  does not require bootloader unlcok ,   just need to download the right files    for dload mode in recovery . and there you go

g-a-b-r-i-e-l, 2021-07-27:
> Can't downgrade from Android 10, because I cant unlock the bootloader. - Catch 22.

So, connect your phone with a cable to your pc, install hisuite on it, then configure it and connect your phone using hisuite, there sould be a button that says update, click on it, wait and there should be a button that says switch to other version or something, press on it and do what it wants you to do for the downgrade. **WARNING: IT WILL ERASE ALL OF YOUR DATA!!!**

closed 2022-05-26 by SkyEmie