SkyEmie/huawei-honor-unlock-bootloader #40: doen't not work

logan001, 2020-07-05:
i keep getting these messages
```
fastboot: error: Command failed
                                                   FAILED (remote: 'invalid command')
```
when i break the execution this is what i get:

^CTraceback (most recent call last):
  File "C:\huawei-honor-unlock-bootloader-master\unlock.py", line 79, in <module>
    codeOEM = tryUnlockBootloader(checksum)
  File "C:\huawei-honor-unlock-bootloader-master\unlock.py", line 24, in tryUnlockBootloader
    sdrout = str(os.system('fastboot oem unlock '+str(algoOEMcode)))
KeyboardInterrupt

fastboot.exe devices will show my device.
adb devices also show my device.

but the program is not working.
i use adb/fastboo from this git. but still fails to work

i have view 20 8gb ram. phone model PCT-L29

logan001, 2020-07-06:
i think the "oem unlock" has been banned
actually i can't run any oem command

fwSmit, 2020-07-09:
I have the same on my honor 10. I also can't run any oem commands I believe. Is this a new protection from huawei?

logan001, 2020-07-09:
i think it came with update to android 10 and ui [add number here] :D

fwSmit, 2020-07-09:
Yeah I think so too. I should have thought of unlocking the bootloader
earlier. If you come across any other free method of unlocking, please let
me know.

logan001, 2020-07-09:
i don't know how this works but you probably can downgrade your phone with hisuite and unlock the phone. but if you upgrade it again you'll probably lock it again.

fwSmit, 2020-07-09:
Oh thanks I'll give that a try.

closed 2020-08-16 by SkyEmie