SkyEmie/huawei-honor-unlock-bootloader #49: This tool is complete garbage

nocturn9x, 2020-08-06:
The code style for this tool is awful, probably one of the worst I've ever seen. The fact that it has 150+ stars is an offense to the Python community and its readability standards :|. I'll soon make a PR to completely rewrite the code, but in the meantime please consider learning Python properly.

PerhapsSomeone, 2020-08-06:
The only garbage thing here is your attitude. 
This is free software made by someone in their free time. If you want to fix it up, go ahead, that's what open source is for, but don't be an ass towards the person **who made this free for everyone to use.**

David-H-1, 2020-08-08:
If you think this is garbage, don't even use it. Go on and waste your money on paid solutions.

nocturn9x, 2020-08-09:
Oh yeah so open source is now an excuse to develop crappy software that goes against any known standard and that might as well have tons of undefined behaviors, yay!

nocturn9x, 2020-08-09:
If it's open source it doesn't mean I can't criticize it, to be honest this tool is so simple and so poorly written that he might have as well spent his time actually learning the language and its standards. Why do you think that things like PRs require approval from the repo owner? For fun? Because code needs to follow a style and a standard, must be tested, readable, and easy to maintain, all things that here are missing

PerhapsSomeone, 2020-08-10:
Shut up and fix it, otherwise, if you're here for nothing but crying about standards, leave.
These are two simple options. If you know how to fix it, do it.
This is the definition of what open source is about.

PerhapsSomeone, 2020-08-10:
#52 is a perfect example of criticism in an issue done right. 
It's constructive, not insulting and actually offers proper recommendations. You are the antithesis of this.

SkyEmie, 2020-08-16:
@nocturn9x I know that the code is horrible to read, only functional under Windows, and you should not have an unexpected error because the code is not designed to. It's a fuk basic tool I made to unlock during many hours the bootloader of my Honor 5x and 8x several years ago when huawei closed the official website, and I put it on github more recently to send it to a friend, I didn't think it would be used that much, and unfortunately I don't have time to deal with it because of other much bigger projects. (and sorry for my English too)

Incon1188, 2020-09-16:
hmm  izzit it work for international version?

closed 2022-05-26 by SkyEmie