SkyEmie/huawei-honor-unlock-bootloader #52: Many things... (no clickbait, you should probably read this...)

programminghoch10, 2020-08-09:
So after hours of bruteforcing I finally had the time to look into the code myself and I have noticed some major problems, which I have mostly fixed on [my fork](https://github.com/programminghoch10/huawei-honor-bootloader-bruteforce).

Click here to get to my fork:
https://github.com/programminghoch10/huawei-honor-bootloader-bruteforce

## Change Overview

* I have rewritten the entire main function to make it more informative and reliable. This includes:
  * The script now shows the progress in %  
     It does not set the window title anymore, as that only works on windows,  
     instead it prints it normally into the terminal, along with other info. #19  #25 
  * You now have some proper and reasonably well documented configuration variables in the code
  * The script now automatically detects if a device has the reboot protection (reboots after 5 failed attempts) and then automatically applies the workaround #6 #16 #26 #31 
    (This is a slightly better implementation than #16, as it only adds the auto reboot if needed)
  * The script now supports numbers below 1000000000000000 (see more down below) #34 #41 
  * The script now stops if it reaches 17 digits #19 #21 #27 #34 #38 
  * The script is now able to actually analyse the command output!  
    Great news! It now is even able to detect if it was successful or not!  
    _kinda depressing as this could mean it has never worked before... but here's hoping_
    (see more below)
  * The script now informs you about actions it is performing, such as intentionally rebooting into bootloader.
  * The script stops if it encounters unexpected results from the device (configurable)
* I have noticed that the luhn checksum is used incorrectly (see more down below) #19 #21 #22 #25 
* The increment actually is the same everytime, so it now gets calculated beforehand to save processing time #25 
* I have enhanced the README to make it more understandable and informative,  
    also I corrected many spelling and grammar mistakes in the README and in the code
* I have added an extra linux script which automatically reboots any device into the bootloader.  
    This can be used for automation purposes, such as bruteforcing over night and using the device over the day,  
    without killing the script in between. 
    I'll add more information about this sometime later.

As you might have already noticed, I have linked some issues my script might potentially fix.

### numbers below 1000000000000000
As stated in #41 the numbers could also be starting with a zero.
I've added the functionality for this, but im an not sure if this would stop the algorithm from calculating correct codes.
If you want to try with zeros, just set the algoOEMcode variable in my script to zero.

### luhn checksum
 [Wikipedias article about the luhn algorithm](https://en.wikipedia.org/wiki/Luhn_algorithm):
```
The Luhn algorithm or Luhn formula (...) is a simple checksum formula used to validate a variety of identification numbers, such as credit card numbers (and) IMEI numbers (...).
```

The Luhn algorithm is only used to __validate__ IMEI numbers.  
The used function is build to __check__ IMEI numbers.

This means that when an IMEI is entered correctly, the checksum would be __zero__!

In  [L49](https://github.com/SkyEmie/huawei-honor-unlock-bootloader/blob/master/unlock.py#L49),
the checksum is added to some other stuff.
When a valid IMEI is entered, the checksum is zero, which renders the use of the luhn algorithm completely useless.

Seriously, try it out yourself!
Open a python 3 terminal, copy the function definition into it, and enter a __valid__ IMEI.

So I did not remove this function, but I removed it from the unlock code generation and used it to verify the entered IMEI.  
I also fixed the README, 
where it states in the [Summary](https://github.com/SkyEmie/huawei-honor-unlock-bootloader/blob/master/README.md#summary) that it uses the luhn algorithm, 
as this is clearly not correct.

### Why SkyEmie's script can't read 👓 
@SkyEmie used the `os.system()` function to execute the fastboot commands.

This function executes the given command, 
but does __not__ return the output of the command.  
It instead [returns the return value of the command](https://www.geeksforgeeks.org/python-os-system-method).

After she executes the fastboot command on [L24](https://github.com/SkyEmie/huawei-honor-unlock-bootloader/blob/master/unlock.py#L24),
the variable only contains a number.

[Splitting the string](https://github.com/SkyEmie/huawei-honor-unlock-bootloader/blob/master/unlock.py#L25) will only convert the number into an array.

Then, the for loop goes over this array (so only once, using the number),
i will __never__ find the string [success](https://github.com/SkyEmie/huawei-honor-unlock-bootloader/blob/master/unlock.py#L25) and also not [reboot](https://github.com/SkyEmie/huawei-honor-unlock-bootloader/blob/master/unlock.py#L34).

So the entire for loop and checks within will never be executed.

Luckily I have been able to fix this.

## Conclusion
This script probably has never worked properly.
Assuming the algorithm she used (except luhn) actually works,
it might be possible that the script got the correct code, but never noticed.

For me it is also not clear, wether the script should start at `0000000000000000` or `1000000000000000`.
@SkyEmie Can you explain your choice to start at `1000000000000000`.?
@nICEnnnnnnnLee and @grossmaul You both have unlocked an european bootloader with numbers. Are you guys able to provide code and IMEI for me to check wether the script would have worked, if it had started at `0000000000000000`?

Good news though, all the bruteforce progress is not lost, 
as I think the mobile would still have shown the unlock prompt or unlocked anyways if the script got it right.

I will propose a pull request, as soon as I am finished optimizing and enhancing my version of it.

P.S. 
I (only) tested the script on linux, as I am using a raspberry to continuously try IMEIs, even when my PC is off.
Please tell me if there are issues on windows.

viluzi, 2020-08-09:
I believe i provided the IMEI in my issue :)

viluzi, 2020-08-09:
My IMEI was: 860899039793009

eclipsek20, 2020-08-10:
Hi.
When i use your software it does not prompt me for any imei (i entered it into the offical software so that might have saved it) and it also goes in a loop (i think it does):
Rebooting to prevent bootloader from rebooting...
Rebooting into bootloader                          OKAY [  0.023s]
Finished. Total time: 0.026s
Bruteforce is running...
Currently testing code 1000000000000000
Progress: 10.0%

Is this behavor normal?
P.S. Sorry for my poor english

orangpelupa, 2020-08-10:
Your code is incorrect, making it doesnt ask for imei. because you put == 0 and > 0 reversed. the fix 

```
while (checksum != 0):
    if staticimei == 0: 
        imei = int(input('Type IMEI: '))
    if staticimei > 0:
        imei = staticimei
```

programminghoch10, 2020-08-10:
oh yeah i noticed that already yesterday but forgot to push it, its up now

mschabhuettl, 2020-08-12:
> Hi.
> When i use your software it does not prompt me for any imei (i entered it into the offical software so that might have saved it) and it also goes in a loop (it repeats this every time):
> 
> Rebooting to prevent bootloader from rebooting...
> Rebooting into bootloader                          OKAY [  0.023s]
> Finished. Total time: 0.026s
> Bruteforce is running...
> Currently testing code 1000000000000000
> Progress: 10.0%
> 
> (I have only provided a snippet of the log. ask if you what me to give you the full log)
> 
> Is this behavor normal?
> P.S. Sorry for my poor english
> 
> EDIT:
> I managed to fix this by entering my imei in the code. 

Let us know when you're done. Thanks.

eclipsek20, 2020-08-12:
So far I'm on 31% (I started with the zeros).
I will try to keep you posted.

sergeyzapor, 2020-08-13:
Will this script work without rebooting if phone test point was shorted befor? Can we overcome through test point shortening the 4-5 attemts interruption?

programminghoch10, 2020-08-13:
@eclipsek20 @mschabhuettl Guys please remember that my script is not in any way more likely to work.

I have in fact calculated that the script (starting from 0) missed @viluzi 's code by 22.053.676.161

You can decrease the increment by removing the 1024 in the increment.  
Altough it then is 1024 times more likely to hit the code, it will also take 1024 times as long.  
Even without the 1024 factor, the script missed @viluzi 's code by 10.816.830

So to be honest, I don't think this script is in any way even close to be based on Huawei's algorithm, if there even is one.

@sergeyzapor I dont really know about the process and I couldn't test this myself because the test point pad came off when i soldered to it :(  
But from what I heard, the bootloader is not locked in that time, so install TWRP and don't worry about the bootloader being locked anymore.

On my device, the reboot only increases the average per code test time about 2 seconds, so although that is a lot over a long time, it doesn't really matter as you will never hit the code anyways.
With my current speed (6s per try) and with raw bruteforcing (starting from 0 and counting up one each attempt), getting to @viluzi 's code (0003387123584726, which is not even that big of a number as there are still 3 zeros in the front) would have _only_ taken 644.429 years.
Without the reboot this would have dropped to 429.619 years.

So if you're able to short the test point and you are able to temporarily bypass the protection, then install your mod and forget about this sh*t.

Still waiting on @nICEnnnnnnnLee & @grossmaul to provide their code + IMEI for testing.

eclipsek20, 2020-08-13:
@programminghoch10 
I am on 40% and still no code although I think I discovered a pattern. When I was looking for unlock codes (to see an example of how the unlock code looks) I noticed that 99% of the time the unlock code was below 5000000000000000.

martobg10, 2020-08-14:
I have Huawei MediaPad T2 7.0 (BGO-DL09)(Android 6.0; EMUI 4.0)(2016 Year)(Europe variant) and don't have success with this formula. I'm now at 76.91% with @programminghoch10's version. With SkyEmie's version, I'm about the same percentages.
This is an old tablet, I guess it's easier to unlock than newer models. Can you help me with anything? Thanks! 🙂

programminghoch10, 2020-08-15:
@martobg10 I do not know anything about huawei bootloaders and/or algorithms.
I found this repository, just as you, and thought it could use some aesthetic and functional changes.

You do not need to try both, mine and @SkyEmie 's version as there have been absolutely zero changes in the algorithm.
(the same codes are tried)
**Please read my previous posts**, as I have stated this already and also mentioned how to increase your chances.

Nobody knows whether Huawei changed their algorithm and sadly assuming they did or didn't does not help in any case.

So not having success is very usual, I do assume nobody ever had success with this.

@eclipsek20 That would half the amount of codes possible but testing every code would still take ~951.293.759,5 years. (again 6s/try)

martobg10, 2020-08-16:
In this situation, the best way for rooting is to use exploits. Unfortunately, I haven’t found an exploit that works for my old device.

Incineroar, 2020-08-20:
> In this situation, the best way for rooting is to use exploits. Unfortunately, I haven’t found an exploit that works for my old device.

I know a lot of Huawei security bulletins for many devices made mention of CVE-2020-0041, I've wondered if it's possible to use that to deduce or reveal the key (or the process of determining the key) from a temp root environment. Unfortunately, nobody on XDA has looked into it, but I feel it may be the best shot. My P20 Pro is still running a January patch so it's still vulnerable to the exploit thankfully, and I've been keeping it that way so I can keep it able to be temp rooted someday.

azrea, 2020-08-25:
@programminghoch10 @Hinazuki @Incineroar @SkyEmie || Any luck for this phone?
You guys talking here something possibility for very old phones. I'm afraid to run this tool overnight in fastboode mode because i already ran this around 1hr just phone usb plug area seriously hot. the phone working smooth so i dont want to break it.

Device: CUN-U29 CUN-U29C567B134 (CUN-U29.A.RX.V01)
Region: South Asia Mediatek MT6582
Android version: 5.1 EMUI: 3.1
Security patch level: march 5 2017
Kernel version: 3.10 (jslave@WUH1000013817 #1) 

Why you want to unlock bootloeader/root: that version very old and some apps no longer support then install lineageos

Hinazuki, 2020-08-25:
@azrea you can unlock it for 5 bucks if you want, check hcu our DC unlocker, you have pre June 2018 security patch so you're phone have high chance to be compatible with it, or you can join Huawei hax telegram group to unlock it for free with slock exploit, check on xda if you haven't already done it.

azrea, 2020-08-25:
@Hinazuki anyway i'm trying to find any no cost method to unlock.

>or you can join Huawei hax telegram group to unlock it for free with slock exploit

Do they support for mediatek devices? i think only kirin.

>check on xda if you haven't already done it.

?

Hinazuki, 2020-08-25:
@azrea Sorry didn't know it was mediatek, yeah check your device on xda, maybe someone already post a tutorial on how to unlock you phone

azrea, 2020-08-25:
> @azrea Sorry didn't know it was mediatek, yeah check your device on xda, maybe someone already post a tutorial on how to unlock you phone

Couldn't find :(

Hinazuki, 2020-08-25:
@azrea Yeah I checked too and nothing... You still have the paid method if you really want to unlock it

titanx980ti, 2020-09-02:
Got a issue where the command just quits after it reaches 200 counts , i assume that the save file has to do something...

MihaiMoldovanu, 2020-09-04:
I have an old PP9 lite with emui 5.0.3 . i ran the tool against it . Took 4 days to complete 

Rebooting to prevent bootloader from rebooting...
rebooting into bootloader...
OKAY [  0.028s]
finished. total time: 0.078s
OEM Code not found!
< waiting for any device >
rebooting...
finished. total time: 0.077s

just456, 2020-09-05:
@MihaiMoldovanu Hello. I don't want to disappoint you, but there is other way faster method to unlock emui 5 devices. But this tool doesn't require a lot of knowledge about phone unlocking.

Anyways. This is my bootloader unlock code for P10 lite: 5670552397735022

azrea, 2020-09-05:
@just456 can you tell about that ways? or i can contact you DM if necessary.
i'm really looking for unlock very old model emui 3.1 android 5.1

just456, 2020-09-06:
@azrea check this xda thread https://forum.xda-developers.com/p8lite/general/alternative-method-unlocking-bootloader-t3799294
I don't remember if it is exact thread I was looking a long time ago, but it should theoretically work

azrea, 2020-09-07:
> @azrea check this xda thread https://forum.xda-developers.com/p8lite/general/alternative-method-unlocking-bootloader-t3799294
> I don't remember if it is exact thread I was looking a long time ago, but it should theoretically work

@just456 this's my phone specification https://github.com/SkyEmie/huawei-honor-unlock-bootloader/issues/52#issuecomment-680012481
that thread 2018. I have tried past and It cannot root using current kingroot either.

psolyca, 2020-10-12:
For information from Huawei official unlock system before shutdown : 
EVA-L09 (P9) :
```
Code : 4448122213131145
IMEI : 869827021786930
Android 7.0
```
G620S-L01
(to be filled later)

programminghoch10, 2020-11-26:
@psolyca Thank you for your code/IMEI combo, i will note it and analyse it when I have enough combos.

@just456 could you also provide your IMEI for analysis?

@ghost Well you got as good of a chance as with any other phone, so basically none at all. I had a fan blowing air over the phone to prevent it from heating up. 
My recommendation: Use your phone as-is as long as it works, and then switch to anything else than Huawei. 
If you don't care and have the time, maybe try the testpoint method and tell us your if it worked.

@Hinazuki 
> you can also use the testpoint exploit to unlock the bootloader temporarily and replace the recovery with twrp, if you don't care about the phone warranty of course.

Your warranty will be gone as soon as you unlock anyway, it's more about the hassle of opening and maybe losing water protection.

programminghoch10, 2020-11-26:
> > In this situation, the best way for rooting is to use exploits. Unfortunately, I haven’t found an exploit that works for my old device.
> 
> I know a lot of Huawei security bulletins for many devices made mention of [CVE-2020-0041](https://github.com/bluefrostsecurity/CVE-2020-0041/tree/master/lpe), I've wondered if it's possible to use that to deduce or reveal the key (or the process of determining the key) from a temp root environment. Unfortunately, nobody on XDA has looked into it, but I feel it may be the best shot. My P20 Pro is still running a January patch so it's still vulnerable to the exploit thankfully, and I've been keeping it that way so I can keep it able to be temp rooted someday.
> 
> I tried using the old version of this brute force tool and a week later an Android was starting to burn into my screen 😂 I'm hoping eventually the process of generating the codes can be deciphered.
> 
> @programminghoch10 you mentioned using a P20 Lite. Was it able to successfully unlock it?

@Incineroar I have taken a look at this. It seems like only a few devices ever had that bug, as it was discovered pretty quickly and most likely none of the Huawei devices ever had a linux kernel with that bug.
With that probability, it's not worth porting the exploit only for it not working then.

If you find other possible interesting exploits, hit me up!

cverstege, 2020-12-26:
This forum accessed via the internet archive could be very interesting.
https://web.archive.org/web/20180310031522/http://forum.gsmhosting.com/vbb/f1014/request-here-free-huawei-bootloader-password-key-2226929/index6.html
This thread was accessible until mid 2018. There are a lot of IMEI+SN and their corresponding unlock keys available.
This might help when reverse engineering how the unlock code could be generated.
But this could only be for older Huawei Phones and Tablets as this thread was deleted in mid 2018.

cverstege, 2021-01-03:
Getting the bootloader code with a tool called DC Unlocker worked for me.
I know this tool looks kind of shady and paying a stranger 4€ via PayPal mostly results in slim chances of success.
I was willing to take the risk and can confirm that I got a valid code from it. Please check if the tool supports your specific model thoroughly before using it.
I needed to install a legacy version of HiSuite from here https://consumer.huawei.com/en/support/hisuite/ like this:
[image]

I used a VM in order to protect me from anything fishy that might be going on in the background.

I know this information shouldn't be shared, but maybe someone might figure out, how the algorithm works:
MediaPad T2 10.0 Pro (HUAWEI FDR-A01L)
Firmware FDR-A01LC001B010SP26_AMSSJan19201819:03:41
IMEI: 860319030024452
SN: LNS6R16407001475
Bootloader-Code: 2104526981526456
LineageOS 17.1 is up and running by now.

reidevries, 2021-02-01:
> I needed to install a legacy version of HiSuite from here https://consumer.huawei.com/en/support/hisuite/ like this:
How did installing HiSuite help? There doesn't seem to be any way of finding the bootloader code from here

cverstege, 2021-02-01:
> How did installing HiSuite help? There doesn't seem to be any way of finding the bootloader code from here

It does not help using this tool. It was needed for the other program (called DC Unlocker) to work which got me the valid bootloader code. I just wanted to share my success with the other program, but use it at your own risk.

eclipsek20, 2021-02-02:
It's been 6 months and this thread is yet to end. I will not be paying those pieces of shit from dc unlocker (i support open source software not closed source and paid). We have to accept our loss, unless some guy manages to find the algorithm that generates codes. It is a loss for us.

just456, 2021-02-03:
> 
> @just456 could you also provide your IMEI for analysis?
> 

Here it is. I hope I'm not too late :D 
Huawei P10 lite IMEI: 863313039621233

If you need other details such as serial, or any other ask. (I no longer use this phone, shuts off, heats up to 60C, slow even after factory reset, sum ting wong with motherboard)

just456, 2021-02-03:
> 
> 
> It's been 6 months and this thread is yet to end. I will not be paying those pieces of shit from dc unlocker (i support open source software not closed source and paid). We have to accept our loss, unless some guy manages to find the algorithm that generates codes. It is a loss for us.

dc unlocker works with emui 5. tried it on oldest emui 8 didnt work. wasted 4€ on 2017 huawei p8 lite

just456, 2021-02-03:
Farmed from:
> https://web.archive.org/web/20180310031522/http://forum.gsmhosting.com/vbb/f1014/request-here-free-huawei-bootloader-password-key-2226929/index6.html

@programminghoch10 

Huawei ALE-L21
IMEI: 869856021194571
SN : W3D7N15C20000448
code: 5439116597170225

Honor FRD-L09
IMEI1: 863691034168908
IMEI2: 863691034188914
SN: 73QDU17207000399
code: 2282629241731615

Honor KIW-l21
imei1 : 863606031444169
imei2 : 863606031464175
sn : W6HDU16B29001803
code: 2473952411050167

Huawei KII-L21
IMEI:  869254027852667
SN: 86MDU16601005128
code: 1201713282202281 

Huawei VNS-L31
IMEI1: 862189038158042
IMEI2: 862189038188056
SN: FUH7N16830003424
code: 1075820469115161

Honor H60-L04
IMEI: 865520021182200
SN: X9LDU14A30000393
code: 2384520513216324

Huawei LON-L29
IMEI1: 861955032600259
IMEI2: 861955032605381
SN: L5R0217307001452
code: 0A4B6OX736OM2R25

Huawei MHA-L29
IMEI1:861957033254375
IMEI:861957033234369
SN:AHK0216B29001893
code: 2YEATBBEIUQ6M6K4

Honor 8S
IMEI:866108021212950
SN:Z7QDU15922005081
code: 6867311363534025

Huawei ALE-L21
IMEI: 860938033727900
SN: W3D7N16302002195
code: 1912111886218111

Huawei Y6II
IMEI1:863128033409702
IMEI2:8631280334450714
SN:LHTDU16924009241
code: 1721077921014416

Huawei TIT-AL00
imei1: 862932036497692
imei2: 862932038869765
sn: AVY9KA96C2603667
code: 6053146325974305

Thomas131, 2021-07-28:
Huawei ALE-L21
IMEI: 867811023198106
IMEI SV: 8678110231981071
SN: QLF7N15806002597
code: 7757177262211019

Thomas131, 2021-07-28:
listed as CSV as well as added a few found in other sources:
Code;IMEI1;IMEI2;SN;Model
```csv
1912111886218111;860938033727900;;W3D7N16302002195;ALE-L21
5439116597170225;869856021194571;;W3D7N15C20000448;ALE-L21
7757177262211019;867811023198106;;QLF7N15806002597;ALE-L21
2282629241731615;863691034168908;863691034188914;73QDU17207000399;FRD-L09
2473952411050167;863606031444169;863606031464175;W6HDU16B29001803;KIW-L21
1201713282202281;869254027852667;;86MDU16601005128;KII-L21
1075820469115161;862189038158042;862189038188056;FUH7N16830003424;VNS-L31
2384520513216324;865520021182200;;X9LDU14A30000393;H60-L04
6867311363534025;866108021212950;;Z7QDU15922005081;Honor 8S
1721077921014416;863128033409702;8631280334450714;LHTDU16924009241;Y6II
6053146325974305;862932036497692;862932038869765;AVY9KA96C2603667;TIT-AL00
4448122213131145;869827021786930;;;EVA-L09
5670552397735022;863313039621233;;;P10 Lite
2104526981526456;860319030024452;;LNS6R16407001475;FDR-A01L
48228324........;;;;CHM-TL00H
2332342218011319;;;;
```

Seems to be using a different algo:
```
0A4B6OX736OM2R25;861955032600259;861955032605381;L5R0217307001452;LON-L29
2YEATBBEIUQ6M6K4;861957033254375;861957033234369;AHK0216B29001893;MHA-L29
```

We need more samples. Still I tried some analysis on the purely numeric ones, analysing every digit on its own, independent of other positions. The individual characters seem to lean to the lower end (a lot of 1 or 2) - for whatever reason.

1st&6th&15th digit: >50% of the numbers are either 1 ore 2; values > 7 rare
2nd digit: 5-6 rare
3rd digit: nothing special
4th digit: lots of 2s
5th digit; nearly 33% of the digits are 1
7th digit: lots of 2s again, 3-5 rare
8th digit: no value <2 found; nearly 33% of values are 2; other 33% are 3&4; 6-8 rare
9th digit: >25% 1s; 3-5 rare again; 40% are 6 or 8; with exception of 1 and 9, no odd numbers found
10th digit: 25% 1s; >50% of values within [1;3]
11th digit: 66% within [0;2]; values 3,4,6,8 missing
12th digit: >75% within [1;3]; values 4,6,8,9 missing
13th digit: >25% 1s; no values > 6 found
14th digit: >25% 1s; no values > 4 found
16th digit: 33% are a 5; 2-4&7-8 rare

8th-10th digit: no 0 found

Slluxx, 2022-02-09:
I wonder what the algorithm to count up is based on. If its skips like 10000 numbers each step, its no wonder that no one had luck before. I am thinking that statistically maybe 16 random digets each loop might be the way to go instead. Obviously this has flaws too since you cant just store each number you tried and compare that with the number you currently generated. Also the chance is high to just slip by the correct imei by a few digits. on the other hand, its statistically very unlikely that the bootloader code is a number like 2222222222222222 or at the very top or bottom of the spectrum.

Id also like to know if AI could be trained to figure out the correlation between imei and bootloader code. I guess we have a way too small datasample for that though and i personally have no idea how to test this.

After ~4 hours i am at 50% (started at 100...0). once thats done ill try the random approach through my raspberry pi and an even more streamlined python script.

yagdev, 2022-02-15:
Hey guys, any luck so far? And how do I stop my Mate 20 Lite from rebooting after 5 tries?

Incineroar, 2022-02-15:
> Hey guys, any luck so far? And how do I stop my Mate 20 Lite from rebooting after 5 tries?

You'll have to run a `fastboot reboot bootloader` every 4 attempts. It reboots into system after 5 tries if no reboot is performed.

yagdev, 2022-02-15:
> > Hey guys, any luck so far? And how do I stop my Mate 20 Lite from rebooting after 5 tries?
> 
> You'll have to run a `fastboot reboot bootloader` every 4 attempts. It reboots into system after 5 tries if no reboot is performed.

Okay, also, after some time my device just stops being detected and if i go to device manager it shows that the device had errors so Windows disabled it. This is fixed by unplugging it and plugging it back in but is there a software solution?

Also another line of thought, couldn't we theoretically exploit the testpoint, as it allows us to flash partitions?

Incineroar, 2022-02-15:
> Okay, also, after some time my device just stops being detected and if i go to device manager it shows that the device had errors so Windows disabled it. This is fixed by unplugging it and plugging [...]

That could be a issue with the cable you're using or your computer. EMUI only enforces the 5 attempts before a reboot but that's it. And as for testpoint, I think that's been done already, but that still requires opening the phone (Which means you're opening seals and removing your water resistance) and using a paid service like DC Unlocker from what I recall.

yagdev, 2022-02-15:
Yeah I remember that DC Unlocker stuff, and also another thing. I'm thinking of giving in and paying for DC Unlocker 3 day licence. Is there any way I could help by doing that? My device doesn't support code reading, but it supports a direct bootloader unlock. Is there any way to like log DC Unlocker to get a better understanding of how they were able to achieve it?

yagdev, 2022-02-15:
I'm gonna try leaving this script on all night and I'm gonna post back for results tomorrow. Wish me luck...

yagdev, 2022-02-15:
Also here's one from my P Smart (FIG-LX1)
IMEI: 8656890432***** (The Huawei account thing doesn't show the last 5 digits of the IMEI)
Unlock Code: 9493304136520108

Zolax9, 2022-02-19:
Here's one from a Huawei P9 Lite (VNS-L31) I unlocked via Potato-NV.
It's a European model, so the unlock code has alphanumerical characters.

IMEI: 860921034533105
Unlock Code: 2WQHK6PCHHWOV2IJ

yagdev, 2022-02-19:
> Here's one from a Huawei P9 Lite (VNS-L31) I unlocked via Potato-NV.
> 
> It's a European model, so the unlock code has alphanumerical characters.
> 
> 
> 
> IMEI: 860921034533105
> 
> Unlock Code: 2WQHK6PCHHWOV2IJ

Im pretty sure that Potato-NV overwrites the actual code so that won't be useful. I was able to conclude this as both of my phones that I unlocked with DC unlocker (FIG-LX1 and ALE-L21) only contained numbers and those unlocked with Potato-NV (FIG-LX1 and ANE-LX1) contained letters too.

closed 2022-05-26 by SkyEmie