SkyEmie/huawei-honor-unlock-bootloader #54: Random Codes, because that is more likely to work (+alphanumeric support)

programminghoch10, 2020-08-15:
After #52 I decided to switch up the code generation to use randomly generated codes.

This script doesn't seem to work anyways, so why not push your luck.

As this is random, it doesn't need an IMEI.

Try it here:  
https://github.com/programminghoch10/huawei-honor-bootloader-bruteforce/tree/random

This kinda fixes #5 #46.

With this opportunity, I took the chance to add letter support.  
By switching one variable to `True`, the script generates random alphanumeric codes.  
This kinda also fixes #23 #27 #47 #50.

I know this is ridiculous, but what isn't in these times.

leosam82, 2020-08-16:
Hi
Sorry for the basic question but since i am really brand new with python i need some help if you dont mind. 

When i try to launch the tool in windows with the 
C:\(path to file location)> python unlock.py

No matter what i try i get 
SyntaxError: unexpected character after line continuation character

what i do wrong?
>>>

xChr11s, 2020-08-17:
Hey,

thanks for your work.
Bruteforce is wokring fine until some point.
After maybe 30 minutes I'm getting the following error:

(bootloader)  Necessary to unlock FRP!
(bootloader)  Navigate to Developer options, and enable "OEM unlock"!
FAILED (remote: Command not allowed)

The phone is at the same screen like when it worked bruteforcing.
I have to restart the phone then. Any idea how to prevent this ?

Thanks

programminghoch10, 2020-08-17:
yup, you need to downgrade your Android/EMUI version, as the command has been removed in EMUI 10 (for me)

Just downgrade as far as possible and try again.

Also you of course have to enable OEM unlock in the developer options

xChr11s, 2020-08-17:
> @xChr11s RTFM, in the readme exactly your problem is addressed in the FAQ section.
> 
> i had this once, you need to downgrade your Android/EMUI version, as the command has been removed in EMUI 10 (for me)
> 
> Just downgrade as far as possible and try again.
> 
> Also you of course have to enable OEM unlock in the developer options
> 
> @leosam82 use python3
> 
> also i cant tell you if it will work, if unsure expect it to not work

Did you even read my comment? I said it's working fine. The error just appears after maybe 30 minutes or so.
Of course the OEM unlock is enabled in the options, otherwise it wouldn't work the first 30 minutes.

programminghoch10, 2020-08-17:
of course i did, i had this too!
downgrade EMUI and it'll work
which EMUI and android version do you use?

sorry for being harsh, I'm losing my sanity with this Huawei crap

xChr11s, 2020-08-17:
> work

Thanks.
I'm currently on EMUI 10.0.0, Android 10 (Huawei P smart 2019).
I'm downgrading it now and will give feedback :)

The unknownfail feature wasn't working as well, has this to do with EMUI 10 too?
Sorry for my bad english x)
 
It's fine 👍

xChr11s, 2020-08-17:
Oke I got it ...
Some people wrote that the message is normal.

I'm now on EMUI 9.1.0 (Android 9), BUILD Nmbr. 9.1.0.314.

FAILED (remote: check password failed!)
finished. total time: 11.512s

Is this normal? It needs nearly 12 seconds to check? RIP.

programminghoch10, 2020-08-17:
Yep, this is exactly why I gave up on this two days ago... :( 
I only had 6s per try though.
I think it probably does some cryptographic or arithmetic stuff and that's why it takes so long.

I don't know why unknownfail doesn't work, a quick investigation hasn't cleared that up yet.

nocturn9x, 2020-08-20:
Newer huawei bootloader codes are generated cryptographically, which means that there's no pattern to look for. From a purely statistical standpoint, random codes are a "better" bet because the algorithm that's used to sign the devices is designed to spread randomness all over the generated digest, which means that a monotonically incrementing integer is far less likely to be the correct code, unfortunately. As an Honor user, I'm disappointed (I can't seem to be able to downgrade to EMUI 8 to use a couple exploits I know of).

# Some numbers

Let's start with the simple case, a decimal integer of 10 digits. The range of decimal digits is 0-9, 10 possibilities. Multiply that by 10, and you get 100 combinations. But wait! We want the number of **permutations**, because `123` is different from `321`, **order matters**. So that grows our tiny little 100 to 10^10, or 10 billion. and that's without even taking into account that bootloader codes are generally 16 digits long, or that the possibilities for each element of the code bump up to 36 when we take into account for alphanumeric codes, making our enormous 10^10 actually 36^10, gargantuan!

## Conclusion

Even if you could test 100 codes per second (which is fairly irrealistic) that would still take ages, literally thousands of years. Some people may be lucky, it's a random code after all, but it's very unlikely

nocturn9x, 2020-08-24:
Your device must also be running EMUI 8 with the June 2018 security patch or lower, or you'll get banned if you attempt to request a slock with another version that's not compatible. I'm one of the dudes that can generate slocks (I made an userbot in that group) so feel free to join, but read the pinned message carefully

Incon1188, 2020-09-14:
hmmm im using mate 10pro emiu 9 i almost run it around 24hours but its still not get my bl lock code

Rstment25, 2020-10-07:
> Your device must also be running EMUI 8 with the June 2018 security patch or lower, or you'll get banned if you attempt to request a slock with another version that's not compatible. I'm one of the [...]

Hey there !

I bring you some good news , it's possible to downgrade the device to whatever version you want. I also ran into "slocks" method of unlocking bootloader while I was looking into how I can get the unlock code.

If you or your friends are willing I'd appreciate if in return for teaching you on how to downgrade the device to the required security patch you could teach me as well how to make slocks for my device by myself so I don't have to flash god knows what binaries in hopes that the guys that made them are legit.

I don't check this often so here's my discord : Rstment#4572 if you're interested.

sergeyzapor, 2020-10-08:
Testpoint method is better then slock method from the point of ciber security. Though twrp and magisk or supersu can too contain the malicious code.

nocturn9x, 2020-10-09:
@Rstment25 the slock method requires an RSA key to generate them and is kept rigorously private. We use paid methods to downgrade anyway so yeah

Edermg05, 2021-08-07:
> Hi
> I have two questions
> 
> 1. Is this at least theoretically working with the P smart 2019?
> 2. Sorry for the basic question but since i am really brand new with python i need some help if you dont mind.
> 
> When i try to launch the tool in windows from python command window with the below syntax
> C:(path to file location)> python unlock.py
> 
> No matter what i try i get this (copy of entry and return)
> 
> > > > C:\Users\alex\Downloads\Huawei unlocker\huawei-honor-bootloader-bruteforce-random\huawei-honor-bootloader-bruteforce-random> python unlock.py
> > > > File "", line 1
> > > > C:\Users\alex\Downloads\Huawei unlocker\huawei-honor-bootloader-bruteforce-random\huawei-honor-bootloader-bruteforce-random> python unlock.py
> > > > ^
> > > > SyntaxError: unexpected character after line continuation character
> 
> what do i do wrong?
> 
> thank you in advance about any help with that basic question
> 
> > > >

Hi @leosam82 ,
I was wondering if you finally could unlock your huawei's bootloader, and if is worth it, as I'm having some troubles with my p-smart 2019 and the unlock.py python script

closed 2022-05-26 by SkyEmie