SkyEmie/huawei-honor-unlock-bootloader #57: fastboot: error: Command failed

TheFurryHacker123, 2020-10-03:
Bruteforce is running...
Currently testing code 1005762115246341
Progress: 10.06%
                                                   FAILED (remote: 'check password failed!')
**fastboot: error: Command failed**

Rebooting to prevent bootloader from rebooting...
Rebooting into bootloader                          OKAY [  0.025s]
Finished. Total time: 0.030s
Bruteforce is running...
Currently testing code 1005792283388992
Progress: 10.06%

_Is this supposed to happen or I fucked something up?_

Fijxu, 2020-11-09:
Only change `unknownfail     = True` to `unknownfail     = False` in the line 25

programminghoch10, 2021-02-28:
@TheFurryHacker123
Which android / EMUI version?

Nuc1eoN, 2021-11-03:
@programminghoch10 I have the same:
```
FAILED (remote: 'check password failed!')
fastboot: error: Command failed
```

Im on EMUI8. Does it make sense to keep it running or is something wrong?

Fijxu, 2021-11-03:
> @programminghoch10 I have the same:
> 
> ```
> FAILED (remote: 'check password failed!')
> fastboot: error: Command failed
> ```
> 
> Im on EMUI 8.0.0. Does it make sense to keep it running or is something wrong?

Thats wrong, you should check if your device is detected by fastboot with the command `fastboot devices` and then execute this script

programminghoch10, 2021-11-03:
@Nux1eoN This is entirely fine the way it is, this is the message that of the bootloader that the key is wrong. Which is obviously expected if you're testing all the bootloader keys. You can try to continue testing, with your old android version the chances are higher, but I'd suggest looking into alternative unlock methods.
Also @Fijxu please stop giving wrong advice. Unknown fail is false by default and as the script does not stop because of unknown errors there is no need to change it. And obviously the device is detected via Fastboot of Fastboot answers with "wrong key".

Nuc1eoN, 2021-11-04:
@programminghoch10 Thank you! I am gonna continue to run it then.

The only issue I have is that after ~30min I am getting an error like this:
```
FAILED (Status read failed (No such device))
fastboot: error: Command failed

Could not parse output.
Please check the output above yourself.
If you want to disable this feature, switch variable unknownfail to False
```
I suppose I will just set it to false then?

PS: Oh and there isn't by chance a way to disable the vibration on each reboot?

programminghoch10, 2021-11-04:
@Nuc1eoN 
I can't do anything about the issue, it's not a problem with the script, but rather with the device.
You can maybe try adding a `sleep(1)` somewhere in the loop.
Setting `unknownfail` to `false` does not solve any issues, idk why people keep thinking that.
And no, you cannot remove the reboot vibration, it's a bootloader thing, and guess what we are just trying to unlock - the bootloader.

Nuc1eoN, 2021-11-04:
@programminghoch10 Setting `unknownfail` to `False` fixed the issue for me. Now I can run it for 10 hours+ and it will not break. I suppose the script just gives up to early in ormal mode :P

closed 2022-05-26 by SkyEmie