SkyEmie/huawei-honor-unlock-bootloader #60: How long should this take

GivouDev, 2020-10-14:
Hey Guys,

ive using this script on a Honor View 10 (BKL-L09)....
And the script is running since yesterday at 23:00.....

It started at 10%? Is this normal??
The phone also have brutefoce protection and restarts as it should....but nor at 16:50 its only at 12.96%....so it whould take weeks to complete 100%.......

Any advise?
LG, and sorry for my bad english.

rauldux, 2020-10-16:
I have this script running for 4 days now and I'am at 80 % now. Unfortunately no luck so far. It starts with 10 % yes. The 'check command fails' is a error return from fastboot I think. Just keep it running I would say

breakgimme, 2020-10-25:
> I have this script running for 4 days now and I'am at 80 % now. Unfortunately no luck so far. It starts with 10 % yes. The 'check command fails' is a error return from fastboot I think. Just keep it running I would say
did you manage to unlock the bootloader? i'm at 35% right now and I don't know if I should waste my time or pay DC Unlocker

Bosa17, 2021-03-03:
Did you manage to unlock using this code? I am using honor 9 lite

rauldux, 2021-03-04:
> Did you manage to unlock using this code? I am using honor 9 lite

No, I did not. After three unsuccessful attempts I put this project on hold.

closed 2022-05-26 by SkyEmie