SkyEmie/huawei-honor-unlock-bootloader #61: OEM Code not found!

bagou4502, 2020-10-17:
Hello i try all code but i have OEM Code not found!

What is the problem?
(STF-L09C432)

rauldux, 2020-10-19:
Problem is that, if i've got my math right here and asume a european version (with only decimal numbers in use), there are 10^16 combinations which are 10 billiards of possibilities. because most devices have brutforce protection and this script works around this rebooting every 4 atemps let's say for simplification one try of combinations takes 1 second. This would then be 321502057.6 years to try all possible combinations. What this script does is trying random numbers. Which brings us to your question, the possibility of not finding the correct combination is relatively high. 
Taking into account alphnumerics non case-sensitiv it would be 10^36

deniscaraus, 2020-10-19:
> Problem is that, if i've got my math right here and asume a european version (with only decimal numbers in use), there are 10^16 combinations which are 10 billiards of possibilities. because most [...]

No, this script doesn't seem to try random numbers. From what i can see, it takes a base of 10^15 and increments it by the IMEI's square root times 1024 until it reaches a number greater than 10^16.

rauldux, 2020-10-19:
agree, let's say not totally random, as the creator of the script said here:
https://github.com/SkyEmie/huawei-honor-unlock-bootloader/issues/54#issue-679514583

Manki9000, 2020-10-22:
I got the same OEM Code not found error on my Huawei P10 Lite.

lcapriotti, 2020-10-30:
same with me with EU Honor 5C

Fijxu, 2020-11-09:
I couldn't find the code, what's the difference between the EU and AMERICA version, is there a way to adapt this code for AMERICAN phones?

I tried in my CAM-L03 and I didn't find anything

If it is possible to adapt the code to make a version of AMERICA I can help

closed 2022-05-26 by SkyEmie