SkyEmie/huawei-honor-unlock-bootloader #72: Any solution for tablets with no IMEI number?

potyt, 2021-01-16:
I tried using the Serial Number, but it is alphanumeric, not just numeric. Any other ideas?

I've kicked it off, having hacked it to use an increment of 1, not got a lot of hope that this will finish in my lifetime!

Cicango, 2021-01-27:
Have you looked on the box from tablet, maybe there is a number that could look like an IMEI?

potyt, 2021-01-29:
Can't see anything. Just a serial number, but that doesn't work. Nothing visible in settings on the device either.
Do we actually have any confirmation this works for devices with an IMIEI number anyway?

Cicango, 2021-01-30:
Have u tried to go under settings>about phone/tablet and find a number similar to an IMEI?

programminghoch10, 2021-02-28:
@potyt
You can use random codes, check #54

closed 2022-05-26 by SkyEmie