SkyEmie/huawei-honor-unlock-bootloader #74: Can I run this tool on multiple instances

Bakaji, 2021-01-21:
I want to accelerate finding my OEM code . so I made a modified version of the script by deleting reboot command and choosing starting oem manually as input(), but I'm concerned if my phone arrive to the correct oem then it will be interrupted by another script instance and fail to unlock my phone (even it's the right code).
ex:
console1 found oem5 as the correct oem code
console 2 :testing oem20,oem21....

is console 2 able to raise an error while console1 unlocking the phone that could damage it ?

Bakaji, 2021-01-21:
**mutual exclusion issue**

programminghoch10, 2021-02-28:
No we cannot. Speed is limited by the target device and the device can only process 1 request at a time.

closed 2022-05-26 by SkyEmie