SkyEmie/huawei-honor-unlock-bootloader #79: Stuck on "testing code 10000000000000"

iNiKKo, 2021-02-19:
Hey so i just want to know if it's supposted to look like this or if it means that it doesn't work

[image]

llight3, 2021-03-07:
Should be fixed by running as root in linux

dlgames2013, 2021-03-09:
how to fix it in Windows?

llight3, 2021-03-09:
No clue how to fix it in windows also this way of doing it is basically impossible nowadays have a read of some of the other isuess, your better off using potato NV if that works if u have a newish phone

Naozumi520, 2021-06-02:
I fix it with a really weird way, It is because of the driver didn't install correctly or even installed on ur pc!
to do this u'll need the adb installer:
https://forum.xda-developers.com/attachment.php?attachmentid=4623157&d=1540039037
after unzip it just open the exe installer, type "Y" in all options, then u'll got the driver....... u'll got nothing install!
Then I use 7zip to unzip the exe installer, inside the installer there's a folder call "driver" with a file "android_winusb.inf"
Go to the windows device manager, there should have a device call "HIxxxx" (mine was "HI3650"),  manually update the driver with that "android_winusb.inf" driver, then u'll got the Android Bootloader Interface driver installed!
after that just try the unlock program agin.

Edermg05, 2021-08-07:
> I fix it with a really weird way, It is because of the driver didn't install correctly or even installed on ur pc!
> to do this u'll need the adb installer: [...]

Where is the device supposed to be? Cause I have unzipped everything as you said, and it has worked, but when I go to device manager, I can see a "Android Phone" but nothing similar to "HIxxxx".
I hope you can help me.

Naozumi520, 2021-08-08:
> Where is the device supposed to be? Cause I have unzipped everything as you said, and it has worked, but when I go to device manager, I can see a "Android Phone" but nothing similar to "HIxxxx".
> I [...]

Maybe it's cuz my phone cpu is kirin and your device is qualcomm? .
Basically it's the same. both are under the adb mode.
Can you try to manually install the device with the"android_winusb.inf"?
It should work just fine.
Good luck!

closed 2022-05-26 by SkyEmie