SkyEmie/huawei-honor-unlock-bootloader #81: Linux instructions improvement

llight3, 2021-03-07:
the program should be run as root pls add this to the readme thanks

SethFalco, 2021-04-07:
Why would it need to be run as root?

llight3, 2021-04-07:
because it uses adb and adb needs root because adb needs to access a system file namely the one your android device is mapped to

SethFalco, 2021-04-07:
I'm on Ubuntu 20.04, and could use adb perfectly fine immediately after installing it via `sudo apt install adb`.
Are you certain the problem might not be to do with your method of installation, or specific setup?

llight3, 2021-04-08:
I'm guessing Ubuntu sets up an fstab/udev allowing usb devices to be mounted by users by default. this isn't normally the case but i guess a better solution would be to add a trip on permission denied to the code

AnonymousDebug, 2021-12-02:
> I'm guessing Ubuntu sets up an fstab/udev allowing usb devices to be mounted by users by default. this isn't normally the case (It wasn't the case on my arch system) but i guess a better solution would be to add a trip on permission denied to the code.
> 
> apparently fastboot is the culprit not adb but the principal still stands
> 
> https://wiki.archlinux.org/index.php/android#Fastboot see the note in the section above regarding `if you get a no permissions error or execution just hangs with < waiting for any device >`

I use Arch (btw) too but I can use ADB without root privileges, fastboot works fine too. What are you talking about?

I assume you are talking about this bit from the android fastboot arch wiki page:

> If you get a no permissions error or execution just hangs with < waiting for any device > then you need to run fastboot as the root user. Alternatively you can install android-udev or the AUR package android-udev-gitAUR and reconnect your device.

You should already have android-udev or android-udev-git package if you are going to do this, but yea adding this to the FAQ/Troubleshooting won't hurt.
I amma make a PR

closed 2022-05-26 by SkyEmie