SkyEmie/huawei-honor-unlock-bootloader #85: Does it work on Huawei P10 Lite?

MrBlood01, 2021-04-17:
Hello! A question. Could this procedure work on a Huawei P10 Lite or would it just be wasted time?

Nasjoe, 2021-06-17:
I'm trying.
Just change the message output error on line 28 of the script unlock.py :
    failmsg = "remote: password wrong"

I'm on Ubuntu 20.04 Pop Os with adb from standard repository

dsardelic, 2021-07-17:
I've just ran the script on Lubuntu 20.04 x64 with my P10 Lite (WAS-LX1) for the European market. Followed the instructions on the main page. Had to install `adb` and `fastboot` from the standard Ubuntu repo. The script started working immediately, without problems. It ran for full **82 hours**, but sadly without success :( Furthermore, my phone had some sort of brute force protection, so the script had it rebooted every 4 failed attempts. I wish I knew the script wouldn't work before having the phone reboot cca 50000 consequtive times, but what can you do.

closed 2022-05-26 by SkyEmie