SkyEmie/huawei-honor-unlock-bootloader #94: Crash

DAMcraft, 2021-07-27:
[image]
Bruteforce is running...
Currently testing code 1000000000000000
Progress: 10.0%
< waiting for any device >
FAILED (remote: 'invalid command')
fastboot: error: Command failed

Could not parse output.
Please check the output above yourself.
If you want to disable this feature, switch variable unknownfail to False

DAMcraft, 2021-07-27:
[image]
Deactivating the feature gives just a ton of errors

AnonymousDebug, 2021-12-02:
Are you running EMUI 10?

MouseMoosz, 2022-03-02:
Please do **NOT** show your IMEI number in public.

closed 2022-05-26 by SkyEmie