SkyEmie/huawei-honor-unlock-bootloader #96: Huawei P8 Lite 2017 Stuck At Fastboot Screen

FarukOzderim, 2021-07-28:
Hello,
On Huawei P8 Lite 2017 
After I run the program, phone successfully reboots into fastboot in the first try. But it does stop at there.
The program gets stuck at %10 progress and does not continue with anything.

Yukkun, 2021-09-11:
The instructions doesn't mention but in my FRD-L09 (Honor 8) I had to enable from Developer settings the option for OEM unlock and had to do one proper reboot.. Then it started working.

Also if you are using HiSuite, it seems it can also mess with it so you should uninstall it if you have it

xXPinkkiPoni96Xx, 2022-01-06:
I am a bit late, but if you are still messing with the P8, I suggest using Github user mashed-potatoes' PotatoNV. Try searching that up.

closed 2022-05-26 by SkyEmie